michaelkubina/SpotMicroESP32 #1: Short & Long version

Nivek92, 2020-08-12:
Hey Michael,
first I want to say great remix of the original MicroSpot. It's the first bigger project I want to replicate.
Looking through the parts folder I saw that there is a short and long version for "Limb_Ball_Bearing_Mount" and "Limb_Servohorn_Mount". Which one is the correct one / which one should I use?

Besides that is there anything I can help you with in regards to this project?

michaelkubina, 2020-08-12:
Hello Nivek,

i really appreciate that you like my remix! I am glad you ask, because this was the only part that i made in 5 different versions, of which only two made it in this repo.

You should use the "long version", which will result in the legs being fully bend when the robot is in its resting position. The upper and lower leg will be parallel in the end (=).

The "normal version" copies the dimensions of the original design, which results in the legs not being able to fully collapse. This means the legs will always look like an angle brackets in the end (<). Maybe with some force you could damage some parts as well, because theoretically you would be able to "overbend" the joint.

Right now i can't think of anything you could help with specifically. I am working an the assembly guide, and will hopefully soon start designing the mountingplate for all of the electronics. Maybe, when assembling one of your self, you could post your issues, or ideas for improvements and so on here in this bugtracker. I guess this would help...

...will you write about the progress anywhere? Blog, Youtube, Reddit or so? This would be interesting, because i could learn something in return.

Nivek92, 2020-08-13:
Thanks for the quick answer, I had nothing in particular in mind, just wanted to offer a helping hand.

I certainly make sure to share any issues and ideas with you.  

In regards to the programming I want to look at https://spotmicroai.readthedocs.io/en/latest/ first and then decide if I want to implement everything myself and build on that library.

But either way I make sure to document the build process and the programming and share it with you but I still have to decide where to document it. Will let you know as soon as I decided.

michaelkubina, 2020-08-26:
The distinction between long and normal version is now in the 3d printed partlist as an additional note. Furthermore it is mentioned in the assembly guide to go with the long version.

closed 2020-08-26 by michaelkubina