michaelkubina/SpotMicroESP32 #2: Full assembly file pls

yuhao1118, 2020-08-18:
Hi,
Appreciate your hard work! As you have not finish the assembly guide, can I have the full assembly cad file so that I can continue to work on it?

michaelkubina, 2020-08-18:
Hello yuhao, 
thank you very much! And thank you for your offer to help, but its something i would like to finish by myself right now, because i am already all into it. I am still working on the assembly guide - its not abandoned ;-) And actually I released it only 8 days ago (which was also the day i started writing it), with the latest addition being made only two days ago. While writing the assembly guide i also build the CAD of an assembled robot - so i have nothing to share here right now for you. But as soon as the CAD-File is complete, i will share it as well, this is for sure.

so i have to aks for a little patience, the next updates will come soon...

I will keep this issue open, as a reminder for releasing the CAD-file

michaelkubina, 2020-08-30:
Hello,

i have completed the assembly guide for all the 3d printed parts. The assembly of the circuitry is not part of it (yet), because this is still in the planning. I added also the FreeCAD-File of the whole robot, which is as complete as it could be right now, but also very messy, because it was build rather quick and dirty as a blueprint for the images needed for this guide. To tidy it up is not a priority for now, because the usecase for this file is rather tight. But who knows, what other will come up with...

I will close this ticket for now, since the guide is completed and the file released.

closed 2020-08-30 by michaelkubina