michaelkubina/SpotMicroESP32 #9: ST7735 Unable to Source

Annihil8ted, 2020-09-24:
Hello! 

I have on hand a display board:

https://www.amazon.com/Display-Module-ST7735-128x160-STM32/dp/B07BFV69DZ/ref=sxts_sxwds-bia-wc-drs-ajax1_0?cv_ct_cx=ST7735&dchild=1&keywords=ST7735&pd_rd_i=B07BFV69DZ&pd_rd_r=e7229f0a-0145-469d-8454-08fab6d008c2&pd_rd_w=pQd8K&pd_rd_wg=sxD3J&pf_rd_p=037ca9fd-790e-4a16-836b-14da89aed20e&pf_rd_r=76G8X3R0ADY1FK8SD96Q&psc=1&qid=1600975421&refinements=p_85%3A2470955011&rnid=2470954011&sr=1-1-25b07e09-600a-4f0d-816e-b06387f8bcf1

I was unable to find one with out an SD card and unfortunately  this doesn't drop into the existing slot on the chassis shell tail. It's about 2-3mm too short. Is the side facing center or the side closest to the push button was extended about 2-3mm (including mounting holes, this would be able to fit in.

Alternatively, could you specify a link on where to find a ST7735 that's exactly 1.77 and without SD card? Or even better, the one that you have?

Thanks!

michaelkubina, 2020-09-25:
This looks like the one i have: https://www.amazon.com/-/de/dp/B082HTWZR6/ref=sr_1_11?dchild=1&keywords=1.77+tft&qid=1601068721&sr=8-11

But you are right, that it's easier to get those with an sd-card-slot than those without. I bought this specific one just because i had it laying around and i read an article about it a long time ago, which lead to buy it: https://www.hackster.io/ocnarf/cheap-1-77-inch-tft-screen-on-esp32-ebe208

I think i will change the part, to fit the one you mentioned (i already ordered one 1.8" TFT with SD-Card today for myself, should be delivered by tomorrow), mainly because it would be easier for others to buy and it costs the same as the one without an sd-card + plus you could try to utilize the sd-card, if you like to. 

I keep this issue open, till i changed the part - afterwards there will be two versions of the tail. Also it should remid me, to clarify that those small cheap tft's have different distances of their mounting holes...

...i guess this the point where modding starts ;-)

michaelkubina, 2020-09-27:
I made now some adjustments to the longer rear-cover to fit the 1.8" TFT. Will print the mounting-plate (cutout) tomorrow evening i think, to test if the display fits and if there are any issues. If everything works out like i hope i would upload the files afterwards.

Annihil8ted, 2020-09-28:
Awesome, excited to see the mods! Thanks for being so responsive too!

michaelkubina, 2020-09-28:
The print was successfull, had to make just a little adjustment afterwards. You can either test beforehand if your display could be mounted properly with the STL i attached (just a cutout ouf the mounting-place)...or you can use directly the full new rear-cover-piece from the parts-directory.

I also made this display the new default-display for this build. 

[1_8_tft_mounting_test.zip](https://github.com/michaelkubina/SpotMicroESP32/files/5294774/1_8_tft_mounting_test.zip)

I close this ticket within the next couple days or after hearing how it went for you - depending what comes first ;-)

Annihil8ted, 2020-09-29:
I'm printing it right now and will let you know how it fits!

Annihil8ted, 2020-09-30:
I printed the long 1.8 ftf rear cover and my tft screen fits perfectly!

Annihil8ted, 2020-09-30:
I will say however in it's current placement, the SD card cannot be used as it'll hit the walls.

michaelkubina, 2020-09-30:
> I'm not sure how you would be able to place the TFT screen such that the SD is made accessible while not in an awkward placement for the screen.

This was something i thought about as well...we might add a slot so one could use a SD-Card, which would stick out at the bottom about 5-10mm in the current state. Or moving the display a little, which might then look misplaced.

If we want to use an SD-Card as some kind of memory, we can always use the micro-sd slot of the ESP32-CAM - at least i think so. You just would not be able to remove it easily...

Annihil8ted, 2020-09-30:
I would opt for the slot because 1) It's pretty well concealed and 2) It makes swapping cards super simple. Then again that's if you deem it even worthy to invest the time to go back in and modify the stl.

I'm also in a little different boat than you too; I plan on using a raspberry pi which has storage up to whatever card you place in it. I'm not sure if the regular ESP32 has storage

michaelkubina, 2020-10-01:
The ESP32 has 512KB SRAM I think and 4MB memory for flashing. Would need to look it up again, but it not this important now.

I will reopen this issue and modify the file - not immediately but soon. I guess having the option to use the slot/sd-card or not is worth the effort, as long as other can benefit from the mod.

michaelkubina, 2020-10-12:
I have made a slot/hole for the sd-card and moved the screen-pocket and the hole for the pushbutton some tiny amount. It prints good, but your printer should be able to brigde 2cm properly. It does not look weird or anything...it might be fiddly to remove the sd-card/pushing it in...but when you use an sd-card-to-micro-sd-adapter you can just remove the micro-sd-card very nicely and leave the rest in place.

I will close this isse, as the parts are now updated and working properly.

closed 2020-10-12 by michaelkubina