michaelkubina/SpotMicroESP32 #12: Frontal sensor and camera support not mountable

JonathanMortes, 2020-11-12:
The model does not allow the camera mount to fit into the ultrasonic mount. Also the Camera support does not fit correctly as it is limited/blocked by the pins. 
[image]
[image]
[image]

michaelkubina, 2020-11-19:
Hi, concerning the SR04 Ultrasonic Sensors...its a squezing mechanism and it is supposed to remove a little from the wegdes on your cameramount (those on the side, which presss the sonar down). The sonar should not be fully pushed down, when adding the cameramount, but instead should stick out a little...so when you gently press them together you also push your sonars a little bit further down. I had no issues with that. It looks in your final picture, like the sonar is also turned by 180°, so that the clock is on the wrong side and not in its dedicated pocket.

The camera on the other side is weird. I know, that its tight and that there is not much tolerance. But it would not in any way lead to a such big gap. Is it possible, that you use the camera has not the OV2640 Version 2? It has a slightly longer cable...this is the only thing i could think of causing this.

I will flag the camera-mounting-issue as a bug and will look into a possible solution, which does not require a specific cam-version.

michaelkubina, 2020-12-06:
Camera-Mounting Issue solved with last update: https://github.com/michaelkubina/SpotMicroESP32/commit/400f984ac320ee224c66f13d99b1b444cfc76c49

From the looks of it and when simulating the other ESP32-CAM in CAD it results in the same offset as seen in the pictures. I can only believe this was the issue and i updated the camera-parts, to fit both the ESP32-CAM OV2640 Ver.1 and Ver.2

Issue will be closed.

closed 2020-12-06 by michaelkubina