michaelkubina/SpotMicroESP32 #15: (New circuitboard) New power supply too big for holes also mounting places for CV and CC needed

JonathanMortes, 2020-12-15:
I suppose they are part of the pending update. For the measurement units I suggest either belly (though it would be too tight) or in the rear as the removal of the TF screen leaves plenty of space. And maybe swap the small DC unit for the big one in terms of placements for better cable management as the new power supply has a big footprint or at least as close to the side as possible.
[image]
[image]
[image]

michaelkubina, 2020-12-17:
Hello Jonathan,

yes, this is from the pending parts update - switching positions was my idea as well. Especially because the neopixel-ring is mounted with screws, which otherwise would have touched the coild of the buck-convertert. I have now completed the new mountingplate for the SZBK07 buck converter and i will leave it parallel to the XL4016...both are usefull i think - for MG996Rs or when more power is needed.

[image]
[image]

You can find the part here - i like the extended one more for cable management:
* https://github.com/michaelkubina/SpotMicroESP32/blob/master/parts/SpotMicroESP32_parts_v1_0_0/stl/Circuitry_Mountingplate_SZBK07.stl
* https://github.com/michaelkubina/SpotMicroESP32/blob/master/parts/SpotMicroESP32_parts_v1_0_0/stl/Circuitry_Mountingplate_SZBK07_Extended.stl

Better measure beforehand, if your screwholes are 39mm x 53,5mm apart in this module...i often encountered modules, which have some other configurations and thus other dimensions. Maybe not in this one, but definetl with the LM2596.

The current and voltage sensor units are now placed in the modular rear-cover. I have added the updated parts already to the repo about two weeks ago. This was, when the OLED was installed in the top-cover and the TFT was removed completly from the rear.

[image]
[image]
[image]

Maybe you want to check out the new should-joint servomounts as well...they are now reconfigured, so that you route your cables through, preventing them from scrubbing along other edges.

michaelkubina, 2020-12-17:
btw: if you are familiar with freecad, or any other cad, than just open/import the original file for the circuitboard and modify parts of it, if you need special changes (like the hole at your lipo).

Feedback appreciated...wiring might be tricky, because you would need cable-shoes and bend them so that they stand up straight. Proper insulations with shrinking hoses to avoid unwanted short circuits are recommended.

JonathanMortes, 2020-12-17:
Hi Michael,
Thanks for the work and the new parts. Regarding my lipo, I changed it for a bigger one and fits like a charm.

JonathanMortes, 2020-12-17:
Hi Michael,

Thanks for the work and the new parts. Regarding my lipo, I changed it for a bigger one and fits like a charm.

Regarding changes on files, I'm experienced with fusion 360 and Autodesk Inventor, I have not touched FreeCad or found an easy way to import it to them, but I'll give it a go if the SZBK07 module's size isn't a good fit. I was planning regardless on adding a bunch more HW to spot: force sensors on the foot and a couple UltraSound sensors on the sides (maybe belly at an angle) so let me know if you need help with any modeling or if you're interested in such enhancements. (I've yet to decide wether to add the neopixels or not as I haven't found a good one on amazon or seen the effects, maybe if you could post some pics that would be nice).

In terms of the CC/CA sensors. Nice placement. There is a black part or rubber to support one (don't know if its printable or not, just to make my own if so) but I have them already printed. 

Finally, ¿have you printed and tested the should-joint servomounts? I was waiting on any changes before printing them as I am not sure about the fit, but if you've done it I'll give it a go!

michaelkubina, 2020-12-18:
The black rubber is just a shrinking hose, that protects the lines on the pcb from the screws scratching them...i had a issue once with the usb ports on my Spotmicro.

The shoulder joint parts are nice, i already printed and use them. 
[image]

Maybe i export all freecad files as step files as well. This way import should be no issue whatsoever, when there is no native import option for freecad files.

Adding more to the spot is always a good idea...we still have plenty of pins left. When using more sonars (or even just with those two) try the Newping.h library (just google it)...it's  really sophisticated.

JonathanMortes, 2021-01-06:
Hello Michael,

I tested those new parts and they work like a Charm #SavedServoWires. Also thanks for the pics on the mounting places for the sensors. My VCC sensor had a weird M1 screw and had to improvise a support fix but they all fit perfectly on the design.

In other news, I remember you were thinking on a solution for unplugging the battery. I joined the male T plug with the main spot power with a screwed double sided cable joint. In case you ever need to dismount spot's power cables and don't want to desolder the deans plug, it comes quite handy as it makes everything power related (except the main pinned power bus) removable without desoldering.

closed 2021-01-06 by JonathanMortes