michaelkubina/SpotMicroESP32 #16: May I use  TD8320MG?

liwenz, 2020-12-24:
Thanks for your share. I have downed all your git .
I have made a robot dog which parts printed from https://www.thingiverse.com/thing:3445283 and mg995 as option.
I hope to change the servo motor as TD8320 as it's 333HZ and mg995 is 50HZ.
The parts need to be edited.The  freecad is used to see and edit.  I found I could not edit the stl file then google here.
I'm a newbee at freecad. Your design is more professional and freecad files are shared 。
May I use your design to assemble TD8320 and MG995?Or which parts  should be edited?
Your assemble is great. Could I have a print parts list file, eg. leg.stl 4 mean 4 legs need to be  printed? 

[image]

[image]

michaelkubina, 2020-12-26:
Those dimensions are pretty close to the ones from the MG996R (40.3*20*37)...but the ~2mm in height might be a problem. I cant promise it would work without issues...without rubber dampeners it should most likely fit, with rubber dampeners it can be too tight.

You can find a partlist here: https://github.com/michaelkubina/SpotMicroESP32/tree/master/parts/SpotMicroESP32_parts_v1_0_0

Just dont use deprecated parts and just use one circuitry-mountingplate (choose a suitable one). Apart from that, if really necessary, you would need to edit a lot of parts to make your servos fit. It's not only the servo-pockets in those parts, but the corresponding servohorn-mounts and maybe even screwholes might be in the way....

So with this all in mind, i leave this up to you if you want to try those servos with the risk of not being able to mount them properly. Technical drawings and measurements with calipers are always two different values and experimenting with other parts than those mentioned are your responsibility.

closed 2021-01-15 by michaelkubina

liwenz, 2021-01-18:
Thanks for your answer. I focus on robot's walking and circuits recently. After some time, I could try the new freecad  design.

Closed #16<https://github.com/michaelkubina/SpotMicroESP32/issues/16>.