michaelkubina/SpotMicroESP32 #17: Bug: Updated top shoulder m3 nut holes too small

JonathanMortes, 2020-12-31:
Hey, catched this one. The bottom shoulder is fine btw.
[image]g

michaelkubina, 2021-01-06:
Hello Jonathan,

from memory those holes should have the same dimensions in both the top and bottom shoulder block. But memory can be wrong, so i have looked into the FreeCAD-Files and i can confirm the dimesions are exactly the same. I think the issue here is the "elephant foot" effect, when printing (e.g. https://all3dp.com/2/elephant-s-foot-3d-printing-problem-easy-fixes/).

In the bottom part the nutholes are printed later, so they are not the first layer touching the buildplate. But in the top part they are printed with the first layer and thus can be slightly smaller in the first 1-2 layers. The width in both parts is at 5,8mm, but in the top part the first 1-2 layers can be off by a margin due to this effect.

The easiest solution is to scrub the edges of the first layer in the nutholes a little bit - the other option is to get rid of this effect.

JonathanMortes, 2021-01-06:
You're right. I did do those quick fixes but waited for confirmation before closing the issue just in case. And indeed, I started to suffer from elephants foot by progressive misscalibration. I guess it was time to recalibrate after almost a kilogram of PLA printed. Thanks for the suggestion and help, Michael!

closed 2021-01-06 by JonathanMortes