michaelkubina/SpotMicroESP32 #18: Bug: mounting plate relay holes misaligned

JonathanMortes, 2020-12-31:
It seems that in the short version, the relay mounting holes aren't as wide as in the long version making only 2 holes mountable.
[image]

michaelkubina, 2021-01-06:
Hmm, i was not able to find the problem...in the FreeCAD-Sketches all four mounting-plates (both long/short for XL4016 and SZBK07) have the same dimensions for the pins. They are 28mm x 20mm apart, which fits the relay i have got. In your case it looks like its off by a whole 1mm and i most likely think, that the modules sold in different world regions or from different resellers/manufacturers have slightly different dimensions...

but none the less, i would like to look at it again ... so could you please tell me, which STL you are using and what shorter STL you have used that had a better fit?

JonathanMortes, 2021-01-06:
After comparing with the two previous long XL4016 boards and the current short SZBK07 board, It turns out they are indeed all shorter by that 1mm except for one American relay I've got that does fit as it houses a smaller board. It's also the case for the usb mini mounting holes. I've bought everything from standard retailers on EU via Amazon so I guess its regional. But as every other component fits It's those I guess. If I ever need a 4th version I guess I'll move them and provide the files for those with slightly bigger components as me. Nevertheless, thanks for checking it out and I'm really sorry for the confusion and time spent looking this things up. Let me make it up to you by sharing this simple addition to the project: a pin mounting plate, that fits in the board and serves as a DIY version of the double sided pin soldering board for connections. 

[pinSupport.stl.zip](https://github.com/michaelkubina/SpotMicroESP32/files/5778483/pinSupport.stl.zip)
[image]

jalber2010, 2021-01-08:
Good morning, first of all congratulations on the project and thank you very much for the great work you are doing.
I just wanted to comment that I have exactly the same problem that JonathanMortes previously mentioned. As you can see in the photo the screw holes are misaligned, only two match.
In my case I live in Spain but I ordered the relays from aliexpress china.
Just in case I ordered two different brands, a total of 6 relays and all of them are misaligned. Not a big deal, just a comment.
In my case I am using the design from 22 days ago with the XL4016. Circuitry_Mountingplate.stl.

On the other hand, I think it would be very good if the Circuitry_Mountingplate had holes to pass the cables through the lower part and thus leave the wiring on the upper part a little more clear. It's just a suggestion. At the bottom only the battery goes and there is a lot of space to be able to pass the cables and fasten them with cable ties.
[image]

michaelkubina, 2021-01-15:
Hello Jalber2010,

thank you for your feedback. I will add the request of widening the distance for the mounting-pins for the relay at about 1mm to my To-Do and will make those changes in the coming ~2 weeks. It seems, that it effect more users, so it seems like a good idea to do this.

The second request...adding some holes to the bottom for cable-routing...well, this is something, that i wont adress actually. With the LiPo in place, there would not be any access from the shoulders sides as it blocks the way ...and routing cables from above under the plate to route them back up seems not benefitial. If you want changes like these, than altering the FreeCAD models i provided to your needs would be the best way to achieve this. This way you can make all modifications you need ;-)

JonathanMortes, 2021-05-13:
I'm redesigning the circuitry mount a bit to add a multiplexor for the I2C channels as well as an arduino nano for force sensor processing and handling all the I2C signals from the different components. And while I'm at it, I've moved the holes 1mm to fix it and I'm also going to move the connections hub and the USB ports to fit the ones I bought which are slightly smaller and need to be higher.

All in all, here is the same sty /step file with the fix. If anyone is interested in the rest of the mods, let me know!

With this I believe the issue is closed.
(please note STL may be displayed on Cura at 100% but in reality it's 10% so scale it to 1000%)
[ReleFixMountingBoard.zip](https://github.com/michaelkubina/SpotMicroESP32/files/6474951/ReleFixMountingBoard.zip)

closed 2021-05-13 by JonathanMortes