michaelkubina/SpotMicroESP32 #21: Rubber Dampeners Question

AugustDG, 2021-03-08:
Hello all!

I'm a student at Dawson College (in Montreal) and I am creating this robot for one of my courses (super excited to do so too 😄) and I had a question about the utility of the rubber dampeners? I have some that came with the servo motors:

[image]

And I imagine they are some sort of adapters to use with M3 screws (since they don't seem to be able to directly screw into the screwhorn), but I'm not sure what exactly they are and how I would use them? I tried to look it up, but nothing came up on Google... 

Any help would be very appreciated!

Otherwise, thank you very much for all the instructions: they are surprisingly clear to read :)

AugustDG, 2021-03-08:
Just saw the dampeners in the CAD model 😅, now know what they're for!

closed 2021-03-08 by AugustDG

michaelkubina, 2021-03-08:
Thats great, was just going to post a picture. I think they might be referred as dampeners as well...but i am not entirely sure with this. Glad you have found it out by yourself!
I wrote the assembly guide only to realize at a very late stage, that i was missing those in the pictures...added a disclaimer at the beginning for it. Also some steps are for deprecated parts (i left them in there), but the new parts should at least be described via text. If you want, then join the discord and share your progresses! its always intersting to see, how others are doing ;-)