michaelkubina/SpotMicroESP32 #22: Servo Motor Voltage

Seasandwpy, 2021-03-15:
Hi,
the servo motor is powered by 6.5V voltage with PCA9685 board, but the max V+ voltage for PCA9685 board is 6V. Will this be a problem?

AugustDG, 2021-03-15:
Can I ask why are the servo motors (or where you saw the servo motors) being powered by 6.5V?

Because if you feed more voltage than it's capable, you'll most definitely damage it some way or another...

michaelkubina, 2021-03-15:
Hello @Seasandwpy and @AugustDG,

it is true, that the Servos (MG996R) are powered at 6,5V via the XL4016/SZBK07 Buck-Converters output (passing through a relay if installed). This is because it provides some additional torque+speed and they are capable of handling this voltage.

@AugustDG 
Its within specs, even if most sellers talk the rated maximum voltage down for reasons i cant tell. Maybe because they only looked at the stall torque of 11KG*cm at 6.0V in those descriptions. The same applies to the rotation speed, which is described at 4.8V and 6.0V.

If you look at the specs at the TowerPro Homepage, its stated that the maximum operational voltage range is from ~4.8V to 6.6V - [look here](https://www.towerpro.com.tw/product/mg996r/)

@Seasandwpy 
The PCA9685 module comes in different configurations. It states on the bottom side that the maximum voltage is 6.0V for V++. Exceeding by 0.5V is not that much and it works in my build without problems...but this is not a reasonably satisfying answer i presume.

The configurations i mentioned are "with polarity protection diode" (never seen one tbh, mine where withoug) and "with capacitor"
- The diode often blows from what i have heard in the channels and actually is not necessary for safe operations - from what i understand its the current, that causes it and not the voltage.
- the ones with a capacitor are only limited by the capacitors rated maximum voltage - often its a 10V 1000uF capacitor or such. So here you can have problems, that you actually exceed ratings, when its a smaller rated one.

The PCA9685 chip itself is only rated up to 5.5V (iirc), but the circuits are sepearated and there is no connection to V++. So no need to worry here and this is also the reason, why 6V max does not make so much sense to me. The powerlines on the pcb can handle higher voltage and only with high currents you should expects problems.

A common mod is to enhance the power-rails (V++ & GND) from the terminal block to the pins (underside of the pins) of the channels with solid copper wires, so that the module itself can handle the high currents better. But thats basically it...

So, the answer i can give: There are good reasons, why 6.5V should not be problematic. If you are unsure, than simply go with 6V...and if you feel comfortable, then increase with 0.1V steps or so until you reach a (imo) safe max of 6.5V.

AugustDG, 2021-03-15:
Great detailed response and thank you! 😁

JonathanMortes, 2021-06-28:
Hi, I'm using 6.5V on the PCA9685 with no problems found. Many servos use 6V for their optimal power and giving them extra juice (+0,5V) is more than safe. Just provide them with enough current or the robot will stall. I believe this issue should be closed.

liwenz, 2021-08-18:
In PCA9685, there are Vcc and V+.  They are different, V+ is for Servo , maybe 6.5V or 8.4V or other. Vcc is for PCA chip and it should be max V is 6V.
How do you choose your battery?
1.2V x 5 or 6 is good for V, but the amp is not enough. 3.7V x 2 =7.4V and maybe 8.4V , but mg995 is only 6-7.2V. I use two diode to low the voltage.
I prefer to 8.4v servo and it does not need to low voltage. eg cls6336hv(expensie), hv5523mg(some cheap)

mattwilliamson, 2023-06-26:
> In PCA9685, there are Vcc and V+. They are different, V+ is for Servo , maybe 6.5V or 8.4V or other. Vcc is for PCA chip and it should be max V is 6V. How do you choose your battery? 1.2V x 5 or 6 is good for V, but the amp is not enough. 3.7V x 2 =7.4V and maybe 8.4V , but mg995 is only 6-7.2V. I use two diode to low the voltage. I prefer to 8.4v servo and it does not need to low voltage. eg cls6336hv(expensive), hv5523mg(some cheap)

@liwenz What diodes did you use? I ordered some 15A on Amazon, but they only drop .136v each.