michaelkubina/SpotMicroESP32 #28: CAM with TTL

JonathanMortes, 2021-06-19:
Hi Michael, with the current circuit layout, how have you overcomed the programming and usage of the cam. Currently I'm unable to solder any cable to the reset pin and without a jumper wire from GPIO 0 to GND, the board is unprogrammable.
Any help is welcomed

closed 2021-06-19 by JonathanMortes

michaelkubina, 2021-06-20:
Hi, i see that it is only mentioned in the Fritzing diagram and i think you might have closed this issue already, because you have become aware of it. I had researched a while before i actually found a solution for this problem. Because as you say, it would have meant to either route the jumper out of the body or to look for a different way like OTA flashing, once it has been flashed with a suitable firmware for this.

I will leave a link here, from where i got my solution...maybe helpful as a insight as well: https://github.com/raphaelbs/esp32-cam-ai-thinker/issues/32

Usage of the CAM btw is as a separate entity...like having two devices in you wireless network. I like this firmware as a start: https://github.com/easytarget/esp32-cam-webserver/

Goodbye :-)

JonathanMortes, 2021-06-20:
Hi Michael! I did not succeed because of the missing RST wire to GPO0 I could not solder a while back, yesterday I succeeded  and after some rewiring, got it to work. I'm also using the Cam as an independent device although it's still wired to the main ESP32 board (maybe it'll be useful at some point. Thanks for the tips by the way!

This is my current setup:

[circuito_v6_ALFREDog.pdf](https://github.com/michaelkubina/SpotMicroESP32/files/6682053/circuito_v6_ALFREDog.pdf)

I've been adding some extra things for more advanced control. I'm presenting the software and control programs as part of my master's  degree final project and I will propose creating a new branch with you with the updated components, designs and most importantly, software. At least its offering some more alternatives with the same base parts. I've even got an alternative version with a custom ultra sonic 360º radar mounted on top

Thanks again for the feedback and work!