michaelkubina/SpotMicroESP32 #29: Short circuit

JonathanMortes, 2021-06-26:
Hey Michael,

I've been having issues with short circuits with the current schematic. Have you had them? Could you share current wiring in photo? I'm seeing it pop up whenever the rele activates in cycles. (I've had a couple of smoky incidents too. Appreciate your or anyone's insight

michaelkubina, 2021-06-28:
Hi Jonathan,
i had not encountered any short circuits with this schematic. But i have heard of blown PCA9685...or to be more precise of the polarity protection diodes, which some modules have and others dont. And the capacitor might blow when either stressed over spec (normaly 10V 100(0?)uF iirc). But if it blows, its rather because of reversed polarity i would assume...apart from this part, which seems to be a weak spot in many builds (not only spotmicro, but other bots as well), i have not heard of any issues. 

Here are some pictures of my current wiring, i hope they allow for some insight:

[image]
[image]
[image]
[image]

JonathanMortes, 2021-06-28:
Hi Michael, thanks for the answer and pics!  Turns out it was a mix of the PCA and the Step down that also fried 1 ESP and damaged another ADC (although it may be just defective as it still works). The only thing that strikes me is the current and voltage sensor which work great separated from the circuit but not when incorporated. Nevertheless, after fixing that and a whole lot of cables and servo angles. I've got my version working! Added 4 MPUs, an I2C mux, 4 pressure sensors and a RPi ZW for RL control based on 12-point bezier curves for gait. What I strongly suggest is investing on some metal round servo horns as they really tend to break and unless you print PETG or high ABS, the PLA just gets too much backlash to be controlable and simply breaks.

closed 2021-06-28 by JonathanMortes