michaelkubina/SpotMicroESP32 #30: Question on hardware components

liamhan0905, 2022-02-27:
Could you explain what this circular circuit is on the first image? What's it used for? 

Also, what is this 3 pins on the 2nd picture?

[image]
[image]

JonathanMortes, 2022-02-27:
Hi, first is led lights which are deprecated. Second is the lipo 2s
charging cable that runs from the belly mounting point to the back (top
case) for easy charging)

liamhan0905, 2022-02-27:
I see thank you for the clear explanation. Could you also add a diagram of how the microusb is connected to other devices? The schema that you provided does not have this detail. 

Also, how is SZBK07 with Extensions different to the one without extension? I see that it just has two extended "beam" at each end but the mountingplate itself are the same size. So I'm confused what the purpose of this extension is. 

Lastly, do you have a .ipt file of the mounting plate? I am planning to add a raspberry pi in addition to run ROS on it. And I'd like to make a mod. Thank you!

[image]

michaelkubina, 2022-03-02:
The micro-usb has 4-5 pads. GND, VCC, D+, D- and in some cases ID/SENSE (for OTG-Devices). I have cut a micro-usb cable for the esp32 to length and soldered the corresponding wirs to the pads on the micro-usb-pcb. I did the same for the Mini-USB-cable that i cut to length for the serial-flasher for the esp32-cam. Be careful not to short the wires when screwing the screws in. I used shrinking hose to protect them...

The "beams" are the extensions...for the neo-pixel LEDs i mentioned (the underperforming, deprecated underglow). thats basically it...you can use it for cable-management if you like, but you dont need them...otherwise the SZBK07 fits this specific DC-DC converter, which would be better suited when you use stronger servos.

I dont know which software uses the *ipt format, but i have provided all models as FreeCAD-Files, so you can make you own modifications. I believe, i also made a raspberry pi zero mod (no support here from my side, not tested since i dont own one), which is in the mods-folder in this repo -> also check out, if you dont want to use the reinforced shoulder mod, but you would need to print other shells for the legs. https://github.com/michaelkubina/SpotMicroESP32/tree/master/parts/SpotMicroESP32_parts_v1_0_0/mods

If you want to use another raspberry than the zero, then you would need to download FreeCAD and do your changes there. Or you export the models as STEP-Files and import them into your workflow. Maybe other formats are also supported, but since i only design in FreeCAD i have no insight on this.

Best regards!